PaulWieland/ratgdo #192: Step down converter for voltage on obstruction sensors?

optik678, 2023-11-07:
I have an older GDO (made in 1996 : Genie CM7600IC) and I've measured the obstruction sensors at ~11.2VDC when made and ~12.2VDC when blocked. Would a step down buck converter work for this if I converted the voltage down to a safer range for these inputs? If so, what range would be my target?

I have a bunch of them in a drawer from other projects, so I thought I'd ask.

PaulWieland, 2023-11-07:
You need a 50% voltage divider. Try two 3k to 10k ohm resistors.

optik678, 2023-11-08:
So, like this? If ASCII art isn't clear, I can draw in paint or something. * represents ratGDO physical terminals : 

Obstruction+VDC
    \
    Resistor
      \ 
      *BLACK
      /
     Resistor
    /
*Ground/WHT

*RED to open button (dry contact)

kramttocs, 2023-11-13:
In the same boat. Have a buck converter arriving today but unsure how to wire it between the gdo and ratgdo. I am assuming the existing sensors wouldn't work if I feed them 6v so how to get them 11.3v but ratgdo 6v?

PaulWieland, 2023-11-13:
If I could have powered ratgdo off of the user accessible GDO terminals, I would have. There is not enough overhead in the power system to do so.

kramttocs, 2023-11-13:
That's a different thing. Not talking about powering the ratgdo. 
This is trying to satisfy the obstruction sensor 5-7vdc range of the ratgdo when the gdo uses 11.3vdc.

rlowens, 2023-11-13:
As stated above, you want a couple of resistors for a voltage divider to bring the 11.3V down to 5-7V for the ratgdo obstruction sensor line, not a step down converter that is used for powering things. https://ohmslawcalculator.com/voltage-divider-calculator

kramttocs, 2023-11-13:
Thanks. Follow you there but still a bit hung up.
Those resistors would in the black wire between the Obstruction Sensor and the 'to obst sensor' terminal on the ratdgo, right?
Then the 'to gdo' black obst going from the ratgdo to the gdo would also be 5-7v, right? 
Would that not cause problems with the gdo when it's expecting 11.3v?

optik678, 2023-11-13:
I just finished hooking mine up and verified that the input from the obstruction sensor is now 50% of the original voltage.  I just need to figure out how to put it into dry contact mode.

In regards to your voltage question, if you create the mentioned voltage divider, you aren't changing anything as far as the GDO is concerned. You are creating a modified voltage signal to the ratGDO device that is a % of the original. If you use the same ohm resistors on both sides, it will be 50% of the original voltage.

rlowens, 2023-11-13:
You always want to connect the ratgdo in parallel with the GDO and sensors/control panel, not disrupting anything. The "to obst sensor" and "to gdo" terminals are connected together and are just there to help you connect everything without having to put more than one wire in the same terminal (in the usual install).

You can use wire nuts or Wago connectors as needed.

The voltage divider goes between the obstruction sensor black wire connection (which goes to both sensors) and the ratgdo. You need that wire still going unmodified to the GDO as well.

kramttocs, 2023-11-13:
Oh. So the diagram here isn't quite right in regards to the obst sensor wiring for the dry contact openers? That's what I've been starting at :)
https://user-images.githubusercontent.com/4663918/277838851-e338c3bf-4eda-447a-9e79-737aa1a622a0.png

optik678, 2023-11-13:
Probably doesn't apply to most of you but I didn't see this features matrix when I was flashing my devices. I have a really old dry contact GDO and I flashed both to ESPHome. Looks like I'll have to use MQTT for now since ESPHome doesn't yet support dry contact : https://paulwieland.github.io/ratgdo/01_features.html

kramttocs, 2023-11-13:
Applies to me 👍 Been holding off ordering the (3rd) unit for this gdo. The others being +2 are working like champs with esphome.

rlowens, 2023-11-13:
Here's my modified wiring diagram, does this make it clear? @PaulWieland maybe include this version somewhere appropriate.
[image]

kramttocs, 2023-11-13:
Extremely appreciated @rlowens

kramttocs, 2023-11-14:
> I just need to figure out how to put it into dry contact mode.

Guess I assumed it would do that automagically based on the feedback it got. If you find something needs to be configured, please report back.

optik678, 2023-11-15:
> > I just need to figure out how to put it into dry contact mode.
> 
> Guess I assumed it would do that automagically based on the feedback it got. If you find something needs to be configured, please report back.

I have ratgdo v2.0 (white board). I had to reflash the firmware on my chip to the 2.26a beta version with the web flasher to get Dry contact mode (beta) to show up as an option for control protocol.

Remaining step : install chip into ratgdo and test button function. 

Future : There are limit switches on my opener and I need to check whether they go to ground or go high. I want to get those into the ratgdo board as well but for now I may just use it as a glorified relay switch as I don't use this garage door very often.