PaulWieland/ratgdo #193: Acceptable formats for MQTT IP address

ellweber, 2023-11-07:
I have tried both numeric and alpha addresses (250.150.50.5 or my.address.com) with and without a prefix (http://, mqtt:// or none) and the board's website fails and does not show up on a local browser with a Configuration page.  My MQTT server is remote with appropriate port-forwarding.

The same board works with Tasmota S/W and the same MQTT config parameters.  It connects, is controllable and the traffic shows normally in MQTT Explorer.  Tasmota does not seem to want a prefix for the address.

If I un-check the MQTT check-box on the initial Configuration page then the board boots and shows the Configuration page normally at 192.168.xxx.yyy or at ratgdo.local. Any MQTT server entries at this point lead to a loss of web access.  The logs then show endless MQTT connect failures as follows:

### MQTT DISCONNECTED ###
MQTT Last Will Params: 
willTopic: ratGDO/status/availability
willPayload: offline
qos: 1
retain: 0
clean session: 1
MQTT attempts=
473
### MQTT DISCONNECTED ###
MQTT Last Will Params: 
willTopic: ratGDO/status/availability
willPayload: offline
qos: 1
retain: 0
clean session: 1
MQTT attempts=
474
### MQTT DISCONNECTED ###
MQTT Last Will Params: 
willTopic: ratGDO/status/availability
willPayload: offline
qos: 1
retain: 0
clean session: 1
MQTT attempts=
475

Any clues to this mystery would be most appreciated and preserve what's left of my sanity!!!

PaulWieland, 2023-11-07:
Could be an authentication issue. Do you see any connection attempts on your server's log? Mosquitto (for example) has a verbose logger that will show when connections are denied.

ellweber, 2023-11-08:
If I uncheck Enable MQTT it still requires an IP address and broker port to proceed.  Then, with only those entries, I do have website access to ratgdo after it reboots but no visible activity in the MQTT broker logs.

Then, if I leave the Enable MQTT box checked and reboot (still no broker log-in credentials) it no longer has website access to the Configuration page.

If I start over again, reload the S/W and then reenter the IP address, port, credentials and MQTT Topic Prefix with Enable MQTT unchecked it will reboot and still retain website access.  Still there is no sign of activity on the MQTT server logs.  

All addresses and credentials are identical to the ones that work, on this same H/W, loaded with Tasmota and with MQTT Explorer.  I do not see any sign of this device in the broker logs.

Is anyone else trying to use ratgdo with a remote MQTT/Mosquitto broker?

ellweber, 2023-11-20:
See #230 for more clarity...

closed 2023-11-20 by ellweber