PaulWieland/ratgdo #195: Possible bad v2.5 unit

kramttocs, 2023-11-08:
Received two units today. One worked great after flashing esphome
 The wifi connect option came up, connected, and I could browse to the web server page. Integrated with HA and connected to an opener.

Tried the second one and it flashed fine but no wifi options came up and when connecting to it via the flasher it doesn't display the firmware in the popup like the other did.

I them flashed it with your mqtt firmware. Success and a wifi connect option but when viewing the logs it is in a wifi connected loop. It shows connected during that loop and my router shows it with the same IP but I can't hit the web server. The browser spinner keeps resetting - confirming the looping. Additional the flasher doesn't show the firmware for it either like it should.

Can provide screenshots if they would help.

ellweber, 2023-11-08:
Your description of symptoms sound familiar, see #193 below.  The onboard webserver becomes inaccessible from a browser, while the router shows a normal connection.  In my case, this seems to be tied particularly to the MQTT specific ratgdo S/W. I am wondering if there is a timing issue that some boards are more sensitive to than others?

PaulWieland, 2023-11-08:
@kramttocs please send an email to ratgdo@gmail.com

closed 2023-11-08 by PaulWieland