PaulWieland/ratgdo #196: Request: Maintain up-to-date comparison of firmware options

agelwarg, 2023-11-08:
Trying to read through different GitHub issues on each project, I realize it’s a little difficult to know the current state of affairs / differences between running the two firmware choices. I see value in both projects being able to reference a similar matrix.  It seems like you have genuine interest in both being options/choices as I see your activity on the ESPhome side as well…do you think maintaining a simple comparison between the two would be helpful?  I wonder if others have similar questions about why to choose one over the other.  As one example, I’m not sure of the security+ 1.0 status on the ESPhome side.

PaulWieland, 2023-11-08:
Yes I do think maintaining a feature matrix is necessary, but things have evolved quickly and I have been a little busy trying to keep up.

I just recently added the feature matrix to the website but it is only for explaining the differences between control protocols. I have a draft going which attempts to add the features from the other firmwares that are in development. 

https://paulwieland.github.io/ratgdo/01_features.html

agelwarg, 2023-11-08:
Yep, I saw that matrix...it's great.  It's also part of the reason I thought of a firmware comparison matrix.  Not sure how I could help, but if I can, just let me know how.

PaulWieland, 2023-11-08:
I just added the firmwares into the matrix. Its a little busy, but hopefully still readable.

agelwarg, 2023-11-08:
I didn't think it was there when I first looked....and then it was.  Glad you said you "just added" because now I know I'm not crazy (at least in this case).

agelwarg, 2023-11-08:
btw, the + is in the matrix legend as &plus;

[image]

PaulWieland, 2023-11-08:
Also fixed. Sometimes I really hate markdown.

closed 2023-11-10 by PaulWieland