PaulWieland/ratgdo #201: Windows 10 Can't Connect to ratgdo

cheatoman, 2023-11-08:
Plugged my new v2.5 board in to try and flash it. It shows up under **Device Manager** as **Other devices>USB Serial** saying it has no compatible drivers. What am I supposed to install? FAQ says USB to UART drivers, but I don't know which one. FTDI VCP? D2XX? D3XX?

sudeepshakya, 2023-11-08:
Same issue here. It seems the newer 2.5 version boards have iFT232-S16 USB to UART chips. Not sure where to get the driver fir this chip.

cheatoman, 2023-11-09:
Originally I installed new FTDI VCP drivers, and it would connect, but fail to flash.

Upon restarting my desktop, it now flashed!

closed 2023-11-09 by cheatoman

sudeepshakya, 2023-11-09:
@cheatoman could you point me to where you got the drivers?

sudeepshakya, 2023-11-09:
Found it. For everyone else here is the link to the drivers
https://ftdichip.com/wp-content/uploads/2021/08/CDM212364_Setup.zip

cheatoman, 2023-11-09:
Yep, thats the one!