PaulWieland/ratgdo #202: MQTT timeout/disconnect issue

ervwalter, 2023-11-09:
I have a 2.5 device that I just received and flashed with the 2.51 firmware.  

It's mostly working except the MQTT connection is not reliable.  I am using mosquito (via the Home Assistant addon).

The symptom is that every 1 minute, the ratgdo gets disconnected from MQTT and reconnects.  If this happens while a command is being sent/processed, it seems like it disrupts it.

This is what I see in the Mosquitto logs in HA:

```
2023-11-08 20:54:26: New connection from 10.5.62.212:63401 on port 1883.
2023-11-08 20:54:26: New client connected from 10.5.62.212:63401 as ratgdo-casey (p2, c1, k15, u'ratgdo').
2023-11-08 20:55:26: Client ratgdo-casey has exceeded timeout, disconnecting.
2023-11-08 20:55:26: New connection from 10.5.62.212:51138 on port 1883.
2023-11-08 20:55:26: New client connected from 10.5.62.212:51138 as ratgdo-casey (p2, c1, k15, u'ratgdo').
[repeated every ~60 seconds]
```

And in the ratgdo console, basically the same pattern also repeated every 60 seconds:
```
[...]
QUEUE MSG SENT [ratgdo-casey-ratgdo-casey/status/availability] 
online
### MQTT DISCONNECTED ###
Connecting to
MQTT Broker...
MQTT Last Will Params: 
willTopic: ratgdo-casey-ratgdo-casey/status/availability
willPayload: offline
qos: 1
retain: 0
clean session: 1

MQTT CONNECTED

Reading data from
the network...
TOPIC SUBSCRIBED [ratgdo-casey-ratgdo-casey/command/#] 
QUEUE MSG SENT [homeassistant/cover/ratgdo-casey/config] 
{
  "~": "ratgdo-casey-ratgdo-casey",
  "name": "Door",
  "unique_id": "ratgdo-casey_2C:F4:32:0E:0D:E2",
  "availability_topic": "~/status/availability",
  "device_class": "garage",
  "command_topic": "~/command/door",
  "payload_open": "open",
  "payload_close": "close",
  "payload_stop": "stop",
  "state_topic": "~/status/door",
  "device": {
    "name": "ratgdo-casey",
    "identifiers": "ratgdo-casey_2C:F4:32:0E:0D:E2",
    "manufacturer": "Paul Wieland",
    "model": "ratgdo",
    "sw_version": "2.51",
    "configuration_url": "http://10.5.62.212/"
  }
}
QUEUE MSG SENT [homeassistant/light/ratgdo-casey/config] 
{
  "~": "ratgdo-casey-ratgdo-casey",
  "name": "Light",
  "unique_id": "ratgdo-casey_2C:F4:32:0E:0D:E2_light",
  "availability_topic": "~/status/availability",
  "command_topic": "~/command/light",
  "payload_on": "on",
  "payload_off": "off",
  "state_topic": "~/status/light",
  "device": {
    "name": "ratgdo-casey",
    "identifiers": "ratgdo-casey_2C:F4:32:0E:0D:E2",
    "manufacturer": "Paul Wieland",
    "model": "ratgdo",
    "sw_version": "2.51",
    "configuration_url": "http://10.5.62.212/"
  }
}
QUEUE MSG SENT [homeassistant/binary_sensor/ratgdo-casey/config] 
{
  "~": "ratgdo-casey-ratgdo-casey",
  "name": "Obstruction",
  "unique_id": "ratgdo-casey_2C:F4:32:0E:0D:E2_obs",
  "availability_topic": "~/status/availability",
  "device_class": "motion",
  "state_topic": "~/status/obstruction",
  "payload_on": "obstructed",
  "payload_off": "clear",
  "device": {
    "name": "ratgdo-casey",
    "identifiers": "ratgdo-casey_2C:F4:32:0E:0D:E2",
    "manufacturer": "Paul Wieland",
    "model": "ratgdo",
    "sw_version": "2.51",
    "configuration_url": "http://10.5.62.212/"
  }
}
QUEUE MSG SENT [ratgdo-casey-ratgdo-casey/status/availability] 
online
[also repeats every 60 seconds or so...]
```

Note, at the moment, this is just sitting on my desk connected via USB so it's not actively connected to the garage door opener, but I did confirm it did the same thing.

Obviously for some reason mosquitto thinks the ratgdo client is timeing out and is disconnecting it, but I can't figure out why.

Thoughts on what I might try?

ervwalter, 2023-11-09:
As a quick test, I setup a vanilla mosquitto server in a docker container on one of my other virtual machines (default config, anonymous access) and it doesn't have this problem.  So it appears to somehow be specific to the mosquitto server that is provided by the home assistant addon.  I'll see if I can experiment furthur to see what the difference between the two setups are.

PaulWieland, 2023-11-10:
You don't by chance have two devices with the same client ID do you?

ervwalter, 2023-11-10:
I did not, though I did erase and reinstall the same ratgdo multiple times, each time giving it the same client id.  But that ID was unique to that device and never was being used by other devices.

In any case, after bashing my head against it for a while, I am still stumped.  The debug logs clearly showed the PINGs going from the ratgdo and being acknowledged, so there was keepalive traffic.  Ultimately, since it was stable with a _different_ mqtt broker, I just ended up keeping that one and bridging it to the HA broker.  That's fine with me anyway as this was the first actual IoT device I had using mqtt (the existing broker was just being used by zigbee2mqtt).  So now I have a separate broker in my IoT VLAN that keeps the IoT devices from being able to talk to my zigbee stuff.

So my personal problem is gone and so I'll close this.

closed 2023-11-10 by ervwalter