PaulWieland/ratgdo #203: Hypothetical Question...

B1ackCr0, 2023-11-09:
I have a ratgdo on order, and as I was pondering whether to take MyQ off the Wifi, a thought occurred: Could Chamberlain send down a a firmware upgrade to it's openers to essentially break the ability that ratgdo has on its openers? In other words, push a firmware down to render ratgdo useless. I'm not a programmer, hardware engineer, or pretend to know how ratgdo works, but I like the ability for Amazon to deliver into my garage. But not for the opportunity to allow Chamberlain to undermine this system. I'm not looking for a guarantee, just measuring the risk. Thanks!

rlowens, 2023-11-09:
The Security+ 2.0 wired protocol that ratgdo emulates is a 9600 baud serial connection with a rolling security code. All Security+ 2.0 door panels use this protocol and we haven't ever seen any way that they would be updateable. And even if they were updateable, current devices would have to connect enough to receive the update, so we could just watch that and probably learn the new protocol.

I doubt the door panels are updateable at all, so they can't break ratgdo functionality since to the GDO it just looks like the door panel sending commands.

PaulWieland, 2023-11-10:
It's unlikely they can use a firmware update to block ratgdo without also breaking all of their own devices. That being said I have never and would never put my GDO on the internet ;)

B1ackCr0, 2023-11-10:
Thanks for taking the time to reply, I have a better understanding now.

closed 2023-11-10 by B1ackCr0