PaulWieland/ratgdo #204: Order Status

AryasDad, 2023-11-09:
How do I get an update on order status? I do not see any contact information.  I would rather not post my email in this public forum. 

I replied to the order receipt from square but not sure if that goes back to you.

Waffleophagus, 2023-11-12:
My situation is somewhat related, so I’d rather make a comment rather than make a new issue. I ordered two devices, one with the Install device and one on its own. But I only got a receipt for one. I’m in no rush, I know they’re on back order. I just wanna make sure both went through as I do have two separate charges on the card. Again no rush, just nervous only one went through.

mrlime, 2023-11-12:
I would also like to know how to contact Paul or RatCloud about a pending order.  I never got a receipt from Square (or anyone else), but I do see the transaction cleared on my credit card statement.  Square says they can't find any receipt.

PaulWieland, 2023-11-15:
Reply to your receipt please.

closed 2023-11-15 by PaulWieland