PaulWieland/ratgdo #206: My ratgdo's are activating door/lights without input

mattvirlee, 2023-11-09:
I bought two 2.5 boards and installed them per the wiring diagram (red -> red, white -> white, etc). They are both flashed with firmware 2.5. My garage doors are Liftmasters with Security 1.0. Both are set to use the experimental Security 1.0 protocol. Both are exposed via MQTT into Home Assistant. 

This morning I heard my garage open as I was getting ready. I heard it again, closing, seconds later. Rinse and repeat -- it was going up and down after random intervals: sometimes nearly instantly, sometimes after a second or two.

Started looking into the status logs HomeAssistant and found a repeating pattern that had been happening for hours from 5AM - 7AM. 

> matt_garage Door was closed
> 7:17:51 AM - 17 minutes ago
> matt_garage Light turned off
> 7:17:51 AM - 17 minutes ago
> matt_garage Obstruction became unavailable
> 7:17:49 AM - 17 minutes ago
> matt_garage Door became unavailable
> 7:17:49 AM - 17 minutes ago
> matt_garage Light became unavailable
> 7:17:49 AM - 17 minutes ago
> matt_garage Obstruction cleared (no motion detected)
> 7:17:39 AM - 18 minutes ago

I can also see that the logbook reports that the lights were turning on and off all night:

> matt_garage Light turned off
> 4:52:20 AM - 3 hours ago
> matt_garage Light turned on
> 4:48:04 AM - 3 hours ago
> matt_garage Light turned off
> 4:48:03 AM - 3 hours ago
> matt_garage Light turned on
> 4:48:00 AM - 3 hours ago
> matt_garage Light turned off
> 4:48:00 AM - 3 hours ago
> matt_garage Door was opened
> 4:47:51 AM - 3 hours ago
> matt_garage Light turned on
> 4:47:37 AM - 3 hours ago
> matt_garage Door is opening
> 4:47:37 AM - 3 hours ago
> matt_garage Light turned off
> 4:42:43 AM - 3 hours ago
> matt_garage Light turned on
> 4:42:42 AM - 3 hours ago
> matt_garage Light turned off
> 4:10:02 AM - 3 hours ago
> matt_garage Light turned on
> 4:10:02 AM - 3 hours ago
> matt_garage Door was closed
> 3:51:41 AM - 4 hours ago
> matt_garage Door is opening
> 3:51:40 AM - 4 hours ago
> matt_garage Light turned off
> 3:49:13 AM - 4 hours ago
> matt_garage Light turned on
> 3:49:12 AM - 4 hours ago
> matt_garage Door was closed
> 3:32:35 AM - 4 hours ago

My spouse's garage door wasn't near as chatty, but it had a similar issue with the lights:

> tori_garage Light turned off
> 3:48:38 AM - 4 hours ago
> tori_garage Light turned on
> 3:48:38 AM - 4 hours ago
> tori_garage Light turned off
> 2:00:40 AM - 6 hours ago
> tori_garage Light turned on
> 2:00:40 AM - 6 hours ago
> tori_garage Light turned off
> 1:01:36 AM - 7 hours ago
> tori_garage Light turned on
> 1:01:36 AM - 7 hours ago

I'm not sure what to debug or next steps to take. I've unplugged them both from mains power at the moment until I can suss out what to do next.

mattvirlee, 2023-11-09:
Perhaps not worth mentioning, but for posterity: the most recent Door status MQTT message is: `syncing` but I believe that is because my first step was to unplug the opener and try a new ratgdo power supply.

mattvirlee, 2023-11-09:
I wonder if I got a bad board - I erassed and reflashed the (most) problematic board with 2.51, reinstalled it, plugged it in and then plugged in the door opener - it immediately opened the door (about 2 inches), reversed direction, repeated twice more, then opened the door fully. 

Readings in Home assistant at that moment showed this repeating:

> matt_garage Obstruction became unavailable
> 10:16:36 AM - 8 minutes ago
> matt_garage Door became unavailable
> 10:16:36 AM - 8 minutes ago
> matt_garage Light became unavailable
> 10:16:36 AM - 8 minutes ago
> matt_garage Obstruction cleared (no motion detected)
> 10:16:27 AM - 8 minutes ago
> matt_garage Door was closed
> 10:16:27 AM - 8 minutes ago
> matt_garage Light turned off
> 10:16:27 AM - 8 minutes ago
> matt_garage Obstruction became unavailable
> 10:16:25 AM - 8 minutes ago
> matt_garage Door became unavailable
> 10:16:25 AM - 8 minutes ago
> matt_garage Light became unavailable
> 10:16:25 AM - 8 minutes ago
> matt_garage Obstruction cleared (no motion detected)
> 10:16:15 AM - 8 minutes ago
> matt_garage Door was closed
> 10:16:15 AM - 8 minutes ago
> matt_garage Light turned off
> 10:16:15 AM - 8 minutes ago
> matt_garage Obstruction became unavailable
> 10:16:13 AM - 8 minutes ago
> matt_garage Door became unavailable
> 10:16:13 AM - 8 minutes ago
> matt_garage Light became unavailable
> 10:16:13 AM - 8 minutes ago
> matt_garage Obstruction cleared (no motion detected)
> 10:16:03 AM - 8 minutes ago
> matt_garage Door was closed
> 10:16:03 AM - 8 minutes ago
> matt_garage Light turned off
> 10:16:03 AM - 8 minutes ago

I pulled the available logs off the device :

```
Dry Contact: open the door
Dry Contact: open the door
Dry Contact: open the door
Dry Contact: open the door
Dry Contact: open the door
Dry Contact: open the door
Dry Contact: open the door
Dry Contact: open the door
Dry Contact: open the door

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (0):
epc1=0x40218e68 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

>>>stack>>>

ctx: sys
sp: 3fffebb0 end: 3fffffb0 offset: 0190
3fffed40:  0000000e 3ffe8aff 00000000 40100269  
3fffed50:  00000000 00000000 4bc6a7f0 ffffffff  
3fffed60:  0000000e 00000001 00000000 401003ec  
3fffed70:  40100b3e 00000000 401009c9 40100bd0  
3fffed80:  e0035015 00000020 401009c9 ffffffff  
3fffed90:  0000000e 3ffe8aff 000028d8 40100290  
3fffeda0:  0000000e 3ffe8aff 000028d8 00000022  
[... 2213 lines cut ...]
    "configuration_url": "http://192.168.1.10/"
  }
}
QUEUE MSG SENT [homeassistant/binary_sensor/matt_garage/config] 
{
  "~": "garage/matt_garage",
  "name": "Obstruction",
  "unique_id": "matt_garage_8C:AA:B5:69:C4:E0_obs",
  "availability_topic": "~/status/availability",
  "device_class": "motion",
  "state_topic": "~/status/obstruction",
  "payload_on": "obstructed",
  "payload_off": "clear",
  "device": {
    "name": "matt_garage",
    "identifiers": "matt_garage_8C:AA:B5:69:C4:E0",
    "manufacturer": "Paul Wieland",
    "model": "ratgdo",
    "sw_version": "2.51",
    "configuration_url": "http://192.168.1.10/"
  }
}
QUEUE MSG SENT [garage/matt_garage/status/availability] 
online
```

PaulWieland, 2023-11-09:
Can you post a photo of the board and the wiring?

mattvirlee, 2023-11-10:
Hey @PaulWieland i had some spare time today and remembered I had a spare ESP board and some GPIO headers here at home. I flashed my spare ESP with the 2.51 firmware and with the same settings as before and didn’t change any of the opener wiring. Worked like a charm and is stable so far. 

Starting to look like I got a bunk D1 mini with one of my boards.

PaulWieland, 2023-11-10:
@mattvirlee send me an email ratgdo@gmail.com

closed 2023-11-10 by mattvirlee