PaulWieland/ratgdo #207: Toggle to enable/disable the ability to open?

malbertSC, 2023-11-09:
Does anyone have a clever solution to "lock out" the ability to open the garage door?

Use case: I have a projector screen for a golf simulator that drops down in front of my garage door.  I want to prevent the garage from accidentally opening while the screen is lowered because the garage door would roll up right into the screen.  The status of the screen (up/down) is already available and controllable in home assistant.  Right now, I have my opener connected to a smart outlet and I have an automation for starting/stopping the golf sim that turns on/off the opener.  However, this setup prevents me from using the battery backup for my opener.

I saw there's a way to disable external remotes, but I don't think that would work for the wall-mounted button.  Is it possible to spoof an obstruction?  And would that prevent the door opening, or just block it from closing?  Is there a better way of doing this?

rlowens, 2023-11-09:
You could put a relay on the wire to the wall-mounted button: relay off=button disabled. Then when the screen is down you turn off that relay and also lock-out the wireless relays. Could also add a template cover in Home Assistant that only works the door when the other requirements are met and hide the real door controls.

PaulWieland, 2023-11-09:
The obstruction can't be spoofed and besides it only prevents the door from closing. I like @rlowens idea of using a relay to disconnect the wall panel. The lockout will block the wireless remotes.

dbrewer-wdc, 2023-11-14:
If you are trying to lock out the wall button and the remotes, you could put in an automation that triggers stops the door when it goes from closed to opening. I have a similar command if for whatever reason, the garage door opens at night when we're all asleep. With the HA automation, the door only gets an inch off the ground before it stops.

If you have the means to control a relay, you could also put the battery backup on the relay so you shut off both power sources.

Or just reach over and disengage the door from the opener, but where's the fun in that.

closed 2023-11-14 by malbertSC