PaulWieland/ratgdo #208: paulwieland.github.io says Failed to execute 'open' on 'SerialPort': Failed to open serial port

spickmws, 2023-11-09:
Two of the devices were delivered today. The first flash and config was flawless. The second device gets the following error when I try to connect: paulwieland.github.io says Failed to execute 'open' on 'SerialPort': Failed to open serial port

I receive this error on both a Windows machine and a Linux machine.

spickmws, 2023-11-09:
If it matters, there are differences between the boards. The board marked D1 mini works and the other does not.

spickmws, 2023-11-09:
Disregard. I flashed it with something else and was then able to use paulwieland.github.io. Apologize for the inconvenience.

closed 2023-11-09 by spickmws

deep-barot, 2023-11-12:
I am facing the problem. how did you fix it?

spickmws, 2023-11-12:
There has to be an easier way, but I ended finding some documentation on using the Arduino IDE to flash blink to an ESP8266 and followed it to try to rule out a hardware problem. It worked. Then I tried again to flash the ratgdo firmware and there were no issues. Again, this was just me fumbling through the problem and there has to be an easier way. Best of luck!

deep-barot, 2023-11-12:
I found the solution

https://github.com/PaulWieland/ratgdo/issues/223#issuecomment-1806958571