PaulWieland/ratgdo #209: Optional Self-hosted MQTT Broker

brbeaird, 2023-11-09:
Would it be feasible to have an option where the ratgdo board itself could server as an MQTT broker? I am thinking of scenarios where users have no other mqtt devices running and just want to add ratgdo with as few steps as possible. With that option, someone could get the ratgdo setup and then just have their chosen integration system connect to ratgdo as the broker and start listening for mqtt messages - no need to stand up a separate machine to run the broker.

rlowens, 2023-11-09:
For reference, here is a tiny MQTT broker for ESP8266 https://github.com/martin-ger/uMQTTBroker

But I think the upcoming HomeKit firmware is probably a more sensible solution to this question https://github.com/ratgdo/homekit-ratgdo

brbeaird, 2023-11-09:
I do not - I only have SmartThings, which does not have one built-in.

PTracks, 2023-11-14:
> If you've got a HomeAssistant instance, can't you use its Broker? https://www.home-assistant.io/integrations/mqtt/#broker-configuration

I think the MQTT integration in Home Assistant is a client only.  No broker functionality.  I had to spin-up Eclipse Mosquitto in Docker to get ratgdo to work with Home Assistant since the ESPHome firmware does not support Security + 1.0 openers yet.

IDisposable, 2023-11-14:
For MQTT in Home Assistant, goto Add Ons / + ADD-ON STORE / (Search) MQTT / Mosquitto broker / Install

PTracks, 2023-11-14:
> For MQTT in Home Assistant, goto Add Ons / + ADD-ON STORE / (Search) MQTT / Mosquitto broker / Install

Ahh, right.  Add-Ons are only available if you are running the full Home Assistant operating system.  I'm running Home Assistant Core in a Docker Container on my Synology NAS, so Add-Ons are not supported.

https://www.home-assistant.io/faq/ha-vs-hassio/

IDisposable, 2023-11-14:
Ah, gotcha! That's why I have HAOS running in a HyperV VM on my Windows 2022 server (at least until my Home Assistant Green ships). It's VMs all the 3ay down 😁

PaulWieland, 2023-11-15:
> I'm running Home Assistant Core in a Docker Container on my Synology NAS

https://hub.docker.com/_/eclipse-mosquitto

PTracks, 2023-11-15:
> > I'm running Home Assistant Core in a Docker Container on my Synology NAS
> 
> https://hub.docker.com/_/eclipse-mosquitto

Yep, that's what I used and it works great.