PaulWieland/ratgdo #211: Wiring Question

jbeck22, 2023-11-09:
I received my ratgdo board today.
My garage door opener is the Chamberlain belt drive (not sure of model #) with the yellow learn button.

I went through the process of installing ratgdo and got my wifi and mqtt settings configured.  I see the device in HA.

I'm sure it is hard to tell, but I have included 2 pics of my wiring and since the controls aren't doing anything I obviously missed a step somewhere.

The wiring diagram that I used to get this far is [this one](https://user-images.githubusercontent.com/4663918/235453980-04a642fa-a181-4297-b4f3-06e1315e02fa.png)

Any advice would be greatly appreciated...

[image]
[image]

PaulWieland, 2023-11-09:
Is it reporting the obstruction status? Use the passthrough terminals on the ratgdo board so you dont have so many wires in one terminal.

jbeck22, 2023-11-09:
when you say passthrough terminals, you mean these?
[image]

sylvr73, 2023-11-09:
I'm only looking into getting this but I believe what he means is if you look at the 2.5 working diagram, there is a red and black that say "to wall..." and "to obst".  The corresponding existing wires go to these and the corresponding "gdo" connections go to the garage door opener.

Disclaimer: Haven't installed one yet myself.

Please report how it goes!

jbeck22, 2023-11-10:
I figured out part of my issue.  When flashing the board originally I chose the wrong firmware version.  Now I have flashed the correct version, but when I click on the "visit device" NOTHING loads other than the link

[image]

I looked at the logs and I see that the webserver has started...

[image]

I have flashed the firmware about 10 times now, tried different cables, used the "reset device" option on the console screen.  Starting to run out of ideas.

PaulWieland, 2023-11-10:
It looks like its partially loading the web page. That has happened to me a couple of times in testing, but usually a single reload fixes it. It could be packet loss.

PaulWieland, 2023-11-10:
> when you say passthrough terminals, you mean these?

These are the passthrough terminals. The two red and two black terminals on the ratgdo board are common so you another port to connect the accessories with.

[image]

wisteso, 2023-11-11:
@PaulWieland Are the two pass-through terminals optional due to being parallel with the respective counterpart? (even if recommended)

The white 'wire/link' overlaying on the PCB suggests as much, along with your comments on this thread, but it would be good to know for sure.

PaulWieland, 2023-11-11:
Yes they’re optional. They’re not present on the v2 board.