PaulWieland/ratgdo #212: Random Unprompted Dry Contact Open and Dry Contact Close signals

nicholasdewitt, 2023-11-09:
Hi, first off, thank you for such a great solution for local garage door control!

Unfortunately, as a few have also mentioned, I am experiencing random, unprompted opens/closes after wiring the RatGDO 2.5 boards to my two Chamberlain 045ACTs (Yellow learn button).

Monitoring the boards' activity through the ESPHome WebUI, on one board, the 'Dry Contact Open' field will randomly flicker from OFF to ON, briefly, causing the door to open, and on the other board, the 'Dry Contact Close' field will do the same thing, which will cause that door to stop and close midway through opening it.

What could cause this? Are there any known troubleshooting steps that can be followed to try to correct this behavior?

Thanks again!

Ramias1, 2023-11-10:
I was also seeing that, but I just attributed it to when I was pressing the wall controller. Perhaps me pressing the wall controller was unrelated. I have since reverted to the MQTT firmware, and that is not working for me either.  Brand new 2.5 board arrived yesterday.

rlowens, 2023-11-10:
You could try wiring the Dry Contact Open/Close terminals to the 3.3V terminal via 10k resistors and see if that fixes it. If it is reproduceable for you enough to tell.

nicholasdewitt, 2023-11-13:
Thanks for the suggestions so far! Unfortunately, no luck with either method, still. Like in @Ramias1 comment, MQTT Firmware does not work either for me. I've only been able to control the garage door with the ESPHome version of the firmware, and that's where I'll experience these phantom signals being sent to dry contact open/close the door.

Is anyone else still experiencing this?

dbrewer-wdc, 2023-11-14:
I saw this happen the other day and thought it had something to do the ratgdo in wall switch emulation mode and having the wall switch connected; perhaps confusing the GDO...don't know. I was going to put a pull up the open and close switches. I'm not sure what the internal pull up value is, but I might try something stronger than 10k; maybe 4.7k. It only happened once over the last few days, but it is a bit troubling.

nicholasdewitt, 2023-11-15:
Thanks Paul! Things have been working with no issues so far after re-flashing with the 2.51 version of the MQTT firmware, so for my own purposes, we can close this issue, but here is my attached picture of my wiring in case you notice something  that could be of use for anyone with similar issues.

[image]

PaulWieland, 2023-11-15:
@nicholasdewitt thanks for that tidbit. I have to check the esphome version and make sure they put enough deadband for the dry contact closure. The mqtt version requires the dry contact to be closed for 500ms.

closed 2023-11-15 by PaulWieland

CrazyCoder, 2023-11-16:
@PaulWieland Please also fix mqtt firmware in dryContact mode as it still uses 50ms debounce instead of 500ms which is too short:

https://github.com/ratgdo/mqtt-ratgdo/blob/45b92632494e01920306eaacd018141cbac6c337/src/ratgdo.cpp#L298
https://github.com/ratgdo/mqtt-ratgdo/blob/45b92632494e01920306eaacd018141cbac6c337/src/ratgdo.cpp#L276C46-L276C46

Here is the issue: https://github.com/ratgdo/mqtt-ratgdo/issues/3.