PaulWieland/ratgdo #213: RatGDO not publishing the topic for the door lock

jermanoid, 2023-11-10:
I have my RATGDOs hooked up and can control the garage door and lights successfully. I saw that they should be able to report and change the door remote lockout mode but I'm not receiving that in home assistant. I just have the door, light, and obstruction sensor

rlowens, 2023-11-10:
For MQTT firmware you need to create a toggle for the topic. For ESPHome firmware it just shows up as a toggle. See https://github.com/PaulWieland/ratgdo/issues/22#issuecomment-1802469125

jermanoid, 2023-11-10:
Alright cool I got it figured out. It took me a minute to figure out where to put that YAML and how to get it right so I'm going to brief future viewers because the HomeAssistant wiki leaves a lot to be desired with examples and scenarios, since I'm using 2 GDOs. The YAML from the comment above goes into your configuration.yaml.I had to make some syntax changes for my setup. Here was my YAML.

`mqtt:
  lock:
    - name: large_garage Lock
      availability_topic: "ratgdo/large_garage/status/availability"
      command_topic: "ratgdo/large_garage/command/lock"
      state_topic: "ratgdo/large_garage/status/lock"
      payload_lock: "lock"
      payload_unlock: "unlock"
      state_locked: "locked"
      state_unlocked: "unlocked"
    - name: small_garage Lock
      availability_topic: "ratgdo/small_garage/status/availability"
      command_topic: "ratgdo/small_garage/command/lock"
      state_topic: "ratgdo/small_garage/status/lock"
      payload_lock: "lock"
      payload_unlock: "unlock"
      state_locked: "locked"
      state_unlocked: "unlocked"
  binary_sensor:
    - name: large_garage Availability
      state_topic: "ratgdo/large_garage/status/availability"
      payload_on: "online"
      payload_off: "offline"
    - name: small_garage Availability
      state_topic: "ratgdo/small_garage/status/availability"
      payload_on: "online"
      payload_off: "offline"`

You can't just paste it into your configuration.yaml unless you set your MQTT topic prefix on your RATGDO config to "ratgdo/" and the name of the two doors are "small_garage" and "large_garage" like mine. otherwise just adjust for the names and MQTT prefix in the topics above to match what you configured for your RATGDO(s). If you need to see what your topics look like on your network, there is a way to view it in HomeAssistant, but i think MQTT Explorer gives you a better global view of the topics that  are bouncing around on your MQTT network. So you can get an idea of what topic you need to list. 

If you only have one door the syntax would omit the little hyphens in each section (since it's no longer a list). It would look like the comment from the link above that I'll paste here  again