PaulWieland/ratgdo #214: Can't connect to UI after initial configuration, and not showing in MQTT (Home Assistant)

LiamThomas7, 2023-11-10:
After flashing, I am able to "Visit" the device and input the basic settings, such as the MQTT server IP and OTA & Web Config Password. After saving and rebooting, I am no longer able to access the UI, despite "Disable OTA & Webserver Config Access" not being checked.

I can see that it is connected to my router, and I am able to PING it.

I have tried reflashing it several times, and connecting it to different WiFi networks.

Any ideas? Thanks.

LiamThomas7, 2023-11-10:
Seems I may have a similar issue to #193, #195 and possibly #181

ellweber, 2023-11-10:
This is now looking like an epidemic!!! I'm sure Paul is looking at it.

LiamThomas7, 2023-11-10:
I figured it out! When I was doing the initial configuration, I was not adding an MQTT username or password. Turns out, if you are using Mosquitto (MQTT) in Home Assistant, it uses your Home Assistant username and password as the MQTT server username and password by default.
As soon as I added my HA username and password to the fields during the initial ratgdo setup, it connected and I am also able to access to web UI.

closed 2023-11-10 by LiamThomas7