PaulWieland/ratgdo #215: Brand new 2.5 with printing upside down?

Ramias1, 2023-11-10:
Its too late tonight to test this, but I saw this thread:  https://community.home-assistant.io/t/ratgdo-getting-collision-detected-waiting-to-send-packet-many-times-per-second-in-log/628335/6 and then this comment: https://community.home-assistant.io/t/ratgdo-getting-collision-detected-waiting-to-send-packet-many-times-per-second-in-log/628335/9 

and found this diagram showing the wires to the left with USB on the bottom:

[image]

Here is a photo of mine; printing for wires to the left with power on the TOP

[image]

Could my issues with Collision Detected be due to the soldering being upside down?  The photo is how my unit arrived.  Thanks

PaulWieland, 2023-11-10:
The D1 mini is installed upside down. Pull it off the board, rotate it 180 degrees and you will be golden. If I sent it to you that way, I apologize!

closed 2023-11-10 by PaulWieland

Ramias1, 2023-11-10:
Thanks.  Yeah it came this way.  I was about to say "it's soldered on" but alas... I was able to pull it right out :).  Will test first thing tomorrow.

Ramias1, 2023-11-10:
Amazing how easy this was with the board orientation correct!