PaulWieland/ratgdo #216: Incorrect door status when Liftmaster 6500 loses power but RatGDO doesn't

bravehand, 2023-11-10:
I'd like to be able to turn off the power to the garage door opener at night, when out of town, etc.

Using the 2.5 board and v2.51 firmware, when I cut the power to the opener the ratgdo status stays "closed" in Home Assistant.  When power is restored to the opener and it's finished booting up (and makes its beeps) the HA status changes to "Open" (but it's still closed).  If I press the "close" button in HA, the door opens and everything is synced up again.

PaulWieland, 2023-11-10:
I'm assuming its control protocol sec+ 1.0?
You have to wait a minute or a little longer for ratgdo to sync. The wall panel spams data for a while until the status is reported.

bravehand, 2023-11-10:
Yes, control protocol is set to sec+ 1.0.

That's the strange thing...after waiting a couple minutes the ratgdo updates but it's changing to "Open" while the door is closed the whole time.  It's still "Open" after 15 min so it looks like that initial sync isn't getting the correct status.

PaulWieland, 2023-11-10:
Okay would you mind capturing and sharing a serial log from startup until the door status syncs?

bravehand, 2023-11-10:
Sure, here the log from power restored till status update:

3131 3131 Obstruction status clear
QUEUE MSG SENT [/home/garage/GarageDoorRatgdo/status/obstruction] 
clear
3131 3131 3131 3131 3131 3131 3131 3131 3131 3131 3131 3131 3131 3131 3131 3131 3131 3131 3131 3131 3131 3131 3131 3131 3131 3131 3131 3131 3131 3131 3131 3131 3131 3131 3131 3131 3131 3131 3131 3131 3131 3131 3131 3131 3131 3131 3131 3131 3131 3131 3131 3131 3131 3131 3131 3131 3131 3131 3131 3131 3131 3131 3131 3131 3131 3131 3131 3131 3131 3131 3131 3131 3131 3131 3131 3131 3131 3131 3131 3131 3131 3131 3131 3131 3131 3131 3131 3131 3131 3131 3131 3131 3131 3131 3131 3131 3131 3131 3131 3131 3131 3131 3131 3131 3131 3131 3131 3131 3131 3131 3131 3131 3131 3131 3131 3131 3131 3131 Looking for security+ 1.0 wall panel...
3131 3131 3131 3131 3131 3131 3131 3131 3131 3131 Looking for security+ 1.0 wall panel...
3131 3131 3131 3131 3131 3131 3131 3131 3131 3131 Looking for security+ 1.0 wall panel...
3131 3131 3131 3131 3131 3131 3131 Looking for security+ 1.0 wall panel...
3131 3535 3333 3800  | door state: 0 0
Door state stopped
QUEUE MSG SENT [/home/garage/GarageDoorRatgdo/status/door] 
stopped
Wall panel detected.
3A5C Light state on
QUEUE MSG SENT [/home/garage/GarageDoorRatgdo/status/light] 
on
3A5C 3A5C 3900 3A5C 3800  | door state: 0 0
3A5C 3900 3A5C 3800  | door state: 0 0
3A5C 3900 3A5C 3800  | door state: 0 0
3A5C 3900 3A5C 3800  | door state: 0 0

gullygossner, 2023-11-16:
I’m having the exact same scenario. It causes me issues as I have automations that will close the door if it is left open. I was caught off guard the other morning when my door was opened because the Ratdgo board thought the door was open and HA called for a door close which actually opened the door. I didn’t experience this behaviour with the v2 board and 2.21 firmware.