PaulWieland/ratgdo #218: Garage door remote working intermittently after installation

scangasn, 2023-11-10:
I purchased and installed 3 ratgdo devices. Flashed all with ESPHome FW and they are all working perfectly via homeassistant ESPHome integration. I noticed after install, that my "dumb" garage door openers seem to be struggling with one of the 3 openers that I installed. I have to click it about 3-4 times before the door responds and finally closes. The other two are working normally. I've doubled checked the wiring and everything is looking fine. Before I remove the ratgdo and check if the button function returns to normal, I wanted to check if anyone has experienced the same?

PaulWieland, 2023-11-10:
What model is the wall panel and what control protocol are you using?

scangasn, 2023-11-10:
Wall panel is [Model 893LM](https://www.liftmaster.com/893lm-garage-door-clicker/p/G893LMMC) and opener is [1D7675](https://www.liftmaster.com/receiver-logic-board-2-0/p/041D7675) which I'm 99% sure is using Security+2.0. I did just confirm that the button in my car is still working properly so it appears to be isolated to the 893LM clicker.

PaulWieland, 2023-11-10:
so if I got you right, you are having an issue with the wireless remote and not the wall panel?

scangasn, 2023-11-10:
ah yes, sorry about that. confused "wall panel" with the "remote". wall panel is working fine and built-in clicker in my car is working fine. only issue seems to be the wireless remote which is still working consistently for my other two doors with ratgdo installed.

scangasn, 2023-11-10:
so i am actually noticing a bit of intermittence on a second garage door opener which is located about the same distance from the remote. I replaced the battery on the remote just to cross that off the list and that didn't help. I may be way off, but I'm wondering if maybe cramming another radio (from the ratgdo) so close to the openers is causing some sort of interference.

PaulWieland, 2023-11-10:
ratgdo uses wifi which operates on 2.4ghz. the wireless remotes are in the 315mhz range. Others have reported a similar issue and it seems as though some power adapters can actually generate interference. maybe try swapping your power adapters around or trying a different one.

scangasn, 2023-11-11:
ok ill try that tomorrow and report back. by the way, brilliant product!

scangasn, 2023-11-11:
@PaulWieland - swapping out the adapters seems to have worked. I was using two old Apple USB adapters and swapped them out for some other ones i had lying around. So far no intermittence, so I will close this issue. Thanks for the quick responses and support!

closed 2023-11-11 by scangasn