PaulWieland/ratgdo #219: Status reversed

ccbadd, 2023-11-10:
I just installed my ratgdo and the status of the position is backwords.  Shows Open when Closed and Closed when Open.  How to I fix this?

rlowens, 2023-11-10:
Is your garage door opener installed backwards? https://old.reddit.com/r/HomeImprovement/comments/bwf7bb/chamberlain_garage_door_opener_has_updown_confused/

ccbadd, 2023-11-11:
No but I think I figured it out kind of.  I flashed it for use with ESPHome but it is a Security+ 1.0 device with a purple button.  I reflashed it for MQTT now and I can get the obstruction status but nothing else.  I'm not familiar with MQTT but I took it to be auto discovered in Home Assistant and I do see it as an MQTT device but it doesn't work.  Do you have to enable something in HASS to get it to work outside of MQTT?

ccbadd, 2023-11-11:
Okay, I got it figured out.  I connected back to the esp and changed the protocol to V1.0 and it works now.  I am a bit bummed that I will not be able to use the motion sensor but I can live with this.

closed 2023-11-11 by ccbadd

PaulWieland, 2023-11-11:
If you have the time, check the serial log and see if any specific data is transmitted when you trip the motion sensor. It might, or it might just trigger the light to turn on. I don't have a sec+1.0 motion sensing wall panel so I haven't been able to test this.

ccbadd, 2023-11-11:
I am not familiar with MQTT to so I need to figure out how to view the logs but I'll check it out when I figure it out.

PaulWieland, 2023-11-11:
To get the serial log from the mqtt firmware you have to plug into it with the USB cable and then connect with the installer. After you connect you'll be presented with an option to view logs and console.

[image]