PaulWieland/ratgdo #220: No control for liftmaster 8500

SaurabhK0001, 2023-11-11:
Connected ratgdo 2.5 with ESPHome firmware to my garage door opener, I have no control (door open / close, light etc) from the device interface or home assistant. The status also does not update.

Attached wiring screenshots 
[image]
[image]
[image]
[image]

PaulWieland, 2023-11-11:
See here: https://github.com/PaulWieland/ratgdo/wiki
You need security + 1.0 control protocol which isn't yet supported by the ESP Home firmware.

[From the feature matrix](https://paulwieland.github.io/ratgdo/01_features.html):
[image]

closed 2023-11-11 by PaulWieland

SaurabhK0001, 2023-11-11:
I’ll try with the Mqtt firmware. The manual for the gdo mentions security +2.0 (attached)

Thanks,
Saurabh

PaulWieland, 2023-11-11:
﻿It’s one of the few transition models. It uses sec+2 wireless but sec+1 wireline control.

SaurabhK0001, 2023-11-11:
Thanks, yes it worked with MQTT when I switched protocol to 1.0. Might be helpful if this information in searchable

Thanks,
Saurabh