PaulWieland/ratgdo #221: Motion sensor independent of automatic light panel setting

GitHubGoody, 2023-11-11:
I have an 880LMW Smart Control Panel that has an integrated motion sensor. At first, after installing ratgdo, I didn't know why the motion sensor entity was not tripping until I remembered that I previously disabled the automatic light on the panel. I re-enabled it and the motion sensor entity started indicating as expected.

Is there any way to disable the automatic light, but still have access to the motion sensor status through ratgdo? I'd like to be able to use it for other automations, but do not want the GDO light to come on when it's tripped.

This may be similar to this [issue](https://github.com/PaulWieland/ratgdo/issues/85), but I wasn't sure.

PaulWieland, 2023-11-11:
Unfortunately no. ratgdo doesn't intercept messages, it just acts as another controller.
You can do what I did though and just take the light bulbs out of the GDO unit. I activate the main garage lights based on events (beam tripping after arriving home, or motion upon entering the garage).

GitHubGoody, 2023-11-11:
Dang. Thanks for the reply and creating ratgdo. It's awesome!

I have an [87504-267 opener](https://www.liftmaster.com/secure-view-ultra-quiet-belt-drive-smart-opener-with-camera-led-corner-to-corner-lighting-and-battery-backup/p/87504-267MC) so I'll have to see if I can disable the light which appears to be an integrated LED array, but maybe not.

closed 2023-11-15 by PaulWieland