PaulWieland/ratgdo #222: Unable to control Chamberlain Security + 1.0

ylwmamba, 2023-11-11:
Hi, I tried to install my Ratgdo v2.5 today to my security + 1.0 (purple learn button) garage opener today. I set it up with MQTT as I saw that ESPhome is not compatible with security +1.0 yet. Everything was able to set up and I can discover it in Home Assistant, however I am unable to control the door or light. It does detect any obstruction correctly. I'm not sure what I'm doing wrong. I checked my wiring a few times and even used a second ESP 8266. I've attached some photos of my wiring. On the Home Assistant Logs it keeps showing that the garage is opening even though it is already open. Any suggestions on what to do next?

[image]

[image]
[image]
[image]

PaulWieland, 2023-11-11:
Can you post the serial log? 
It's best to capture it right after restarting it.

[image]

rykm48, 2023-11-11:
I have a similar problem. Just wired it up today!  Using mqtt. I can control the door but am getting a lot of messages in the logbook every second of the door was opened, closed, is closing was closed. Like it’s going nuts with the messages although my door isn’t randomly opening. On occasion I will hear a beep beep beep on my wall button, which the wall button maybe security 2.0 I’m not sure. I bought it originally to connect the myq gateway as the opener itself is a bit older and has a purple button meaning its security 1.0 and the opener itself doesn’t support myq, I think the wall button made that compatible. Should I reflash and change my protocol to security 2.0?

ylwmamba, 2023-11-11:
> Can you post the serial log? It's best to capture it right after restarting it.
> 
> [image]

I just had a chance to pull the logs... It looks like it's not finding the wall panel.

Obstruction status clear
QUEUE MSG SENT [/home/garageRatgdo/status/obstruction] 
clear
Obstruction status obstructed
QUEUE MSG SENT [/home/garageRatgdo/status/obstruction] 
obstructed
rl�lï¿½r$ï¿½nï¿½lï¿½b|ï¿½ï¿½ï¿½rbï¿½bï¿½nnlnnbbpï¿½$blrlpï¿½nï¿½ï¿½lï¿½bnï¿½nï¿½ï¿½bï¿½ï¿½nn'lï¿½l`ï¿½nn$`nrï¿½ï¿½ï¿½nrrï¿½`pï¿½nï¿½rï¿½bbnï¿½nbï¿½ï¿½nn'l`ï¿½nn$`nrï¿½ï¿½ï¿½nrl`rï¿½ï¿½nrl`ï¿½ï¿½$bï¿½l`ï¿½ï¿½nï¿½`
Mounting LittleFS...
[setup.json]
JSON parsed
Storage OK, restoring WiFi and MQTT config.
Connecting to: 
WinstonWifi...
Using static IP address
Set WiFi output power to: 20.50
....
WIFI CONNECTED
IP Address: 192.168.6.46
nb of attempts: 4
IMPROVhttp://192.168.6.46ï¿½

WiFi connected
Local IP: 192.168.6.46
SoftAP IP: (IP unset)
Server started
Webserver ready. Open http://Ratgdo.local/ in your browser
Launching webserver for improv
Starting ArduinoOTA service
doorCommandTopic: /home/garageRatgdo/command/door
lightCommandTopic: /home/garageRatgdo/command/light
lockCommandTopic: /home/garageRatgdo/command/lock
Using security+ 1.0
Setup Complete
 _____ _____ _____ _____ ____  _____ 
| __  |  _  |_   _|   __|    \|     |
|    -|     | | | |  |  |  |  |  |  |
|__|__|__|__| |_| |_____|____/|_____|
version 2.51
IMPROVhttp://192.168.6.46ï¿½
### MQTT DISCONNECTED ###
Connecting to
MQTT Broker...
MQTT Last Will Params: 
willTopic: /home/garageRatgdo/status/availability
willPayload: offline
qos: 1
retain: 0
clean session: 1

MQTT CONNECTED

Reading data from
the network...
TOPIC SUBSCRIBED [/home/garageRatgdo/command/#] 
QUEUE MSG SENT [homeassistant/cover/Ratgdo/config] 
{
  "~": "/home/garageRatgdo",
  "name": "Door",
  "unique_id": "Ratgdo_34:94:54:8E:2B:A9",
  "availability_topic": "~/status/availability",
  "device_class": "garage",
  "command_topic": "~/command/door",
  "payload_open": "open",
  "payload_close": "close",
  "payload_stop": "stop",
  "state_topic": "~/status/door",
  "device": {
    "name": "Ratgdo",
    "identifiers": "Ratgdo_34:94:54:8E:2B:A9",
    "manufacturer": "Paul Wieland",
    "model": "ratgdo",
    "sw_version": "2.51",
    "configuration_url": "http://192.168.6.46/"
  }
}
QUEUE MSG SENT [homeassistant/light/Ratgdo/config] 
{
  "~": "/home/garageRatgdo",
  "name": "Light",
  "unique_id": "Ratgdo_34:94:54:8E:2B:A9_light",
  "availability_topic": "~/status/availability",
  "command_topic": "~/command/light",
  "payload_on": "on",
  "payload_off": "off",
  "state_topic": "~/status/light",
  "device": {
    "name": "Ratgdo",
    "identifiers": "Ratgdo_34:94:54:8E:2B:A9",
    "manufacturer": "Paul Wieland",
    "model": "ratgdo",
    "sw_version": "2.51",
    "configuration_url": "http://192.168.6.46/"
  }
}
QUEUE MSG SENT [homeassistant/binary_sensor/Ratgdo/config] 
{
  "~": "/home/garageRatgdo",
  "name": "Obstruction",
  "unique_id": "Ratgdo_34:94:54:8E:2B:A9_obs",
  "availability_topic": "~/status/availability",
  "device_class": "motion",
  "state_topic": "~/status/obstruction",
  "payload_on": "obstructed",
  "payload_off": "clear",
  "device": {
    "name": "Ratgdo",
    "identifiers": "Ratgdo_34:94:54:8E:2B:A9",
    "manufacturer": "Paul Wieland",
    "model": "ratgdo",
    "sw_version": "2.51",
    "configuration_url": "http://192.168.6.46/"
  }
}
QUEUE MSG SENT [/home/garageRatgdo/status/availability] 
online
Looking for security+ 1.0 wall panel...
Looking for security+ 1.0 wall panel...
Looking for security+ 1.0 wall panel...
Obstruction status clear
QUEUE MSG SENT [/home/garageRatgdo/status/obstruction] 
clear
Obstruction status obstructed
QUEUE MSG SENT [/home/garageRatgdo/status/obstruction] 
obstructed
Looking for security+ 1.0 wall panel...
Obstruction status clear
QUEUE MSG SENT [/home/garageRatgdo/status/obstruction] 
clear
Looking for security+ 1.0 wall panel...
Looking for security+ 1.0 wall panel...
Looking for security+ 1.0 wall panel...
Looking for security+ 1.0 wall panel...
Looking for security+ 1.0 wall panel...
Looking for security+ 1.0 wall panel...
Looking for security+ 1.0 wall panel...
Looking for security+ 1.0 wall panel...
Looking for security+ 1.0 wall panel...
Looking for security+ 1.0 wall panel...
Looking for security+ 1.0 wall panel...
Obstruction status obstructed
QUEUE MSG SENT [/home/garageRatgdo/status/obstruction] 
obstructed
Obstruction status clear
QUEUE MSG SENT [/home/garageRatgdo/status/obstruction] 
clear
Looking for security+ 1.0 wall panel...
Obstruction status obstructed
QUEUE MSG SENT [/home/garageRatgdo/status/obstruction] 
obstructed
Obstruction status clear
QUEUE MSG SENT [/home/garageRatgdo/status/obstruction] 
clear
Looking for security+ 1.0 wall panel...
Looking for security+ 1.0 wall panel...
No wall panel detected. Switching to emulation mode.
3535 33FF 3A39 Light state off
QUEUE MSG SENT [/home/garageRatgdo/status/light] 
off

[image]

This is my wall panel.

PaulWieland, 2023-11-11:
@ylwmamba everything looks normal, although there are a bunch of obstruction detections happening. Was something breaking the beam?

Temporarily unhook your wall panel, restart ratgdo and do another serial capture. This time let it go for longer. It only captured the first set of instructions & responses being sent after emulation mode was started. After emulation mode starts, try and cycle the door and light with ratgdo. Then resend the serial log so I can have a look.

There might be an incompatibility with the way I wrote the emulator and that specific wall button.

ylwmamba, 2023-11-11:
@PaulWieland Thank you for looking into this. Yes in the original log those obstruction detections were correct as my dog was walking back and forth. I followed your instructions and Ratgdo was able to open/close and control the lights but the obstruction sensor was being a bit whacky for a bit (this time detecting obstructions that were not there, I had to wave my hand back and forth a bit before it cleared). Here is the new log:

Mounting LittleFS...
[setup.json]
JSON parsed
Storage OK, restoring WiFi and MQTT config.
Connecting to: 
WinstonWifi...
Using static IP address
Set WiFi output power to: 20.50
....
WIFI CONNECTED
IP Address: 192.168.6.46
nb of attempts: 4
IMPROV      http://192.168.6.46ïŋ―

WiFi connected
Local IP: 192.168.6.46
SoftAP IP: (IP unset)
Server started
Webserver ready. Open http://Ratgdo.local/ in your browser
Launching webserver for improv
Starting ArduinoOTA service
doorCommandTopic: /home/garageRatgdo/command/door
lightCommandTopic: /home/garageRatgdo/command/light
lockCommandTopic: /home/garageRatgdo/command/lock
Using security+ 1.0
Setup Complete
 _____ _____ _____ _____ ____  _____ 
| __  |  _  |_   _|   __|    \|     |
|    -|     | | | |  |  |  |  |  |  |
|__|__|__|__| |_| |_____|____/|_____|
version 2.51
IMPROV      http://192.168.6.46ïŋ―
### MQTT DISCONNECTED ###
Connecting to
MQTT Broker...
MQTT Last Will Params: 
willTopic: /home/garageRatgdo/status/availability
willPayload: offline
qos: 1
retain: 0
clean session: 1

MQTT CONNECTED

Reading data from
the network...
TOPIC SUBSCRIBED [/home/garageRatgdo/command/#] 
QUEUE MSG SENT [homeassistant/cover/Ratgdo/config] 
{
  "~": "/home/garageRatgdo",
  "name": "Door",
  "unique_id": "Ratgdo_34:94:54:8E:2B:A9",
  "availability_topic": "~/status/availability",
  "device_class": "garage",
  "command_topic": "~/command/door",
  "payload_open": "open",
  "payload_close": "close",
  "payload_stop": "stop",
  "state_topic": "~/status/door",
  "device": {
    "name": "Ratgdo",
    "identifiers": "Ratgdo_34:94:54:8E:2B:A9",
    "manufacturer": "Paul Wieland",
    "model": "ratgdo",
    "sw_version": "2.51",
    "configuration_url": "http://192.168.6.46/"
  }
}
[... 479 lines cut ...]
3A55 3900 3801  | door state: 1 1
3A55 3900 3801  | door state: 1 1
3A55 3900 3801  | door state: 1 1
3A55 3900 3801  | door state: 1 1
3A55 3900 3801  | door state: 1 1
3A55 3900 3801  | door state: 1 1
3A55 3900 3801  | door state: 1 1
3A55 3900 3801  | door state: 1 1
3A55 3900 3801  | door state: 1 1
3A55 3900 3801  | door state: 1 1
3A55 3900 3801  | door state: 1 1
3A55 3900 3852  | door state: 52 1010010
Door state open
QUEUE MSG SENT [/home/garageRatgdo/status/door] 
open
3A55 3900 3852  | door state: 52 1010010
3A55 3900 3852  | door state: 52 1010010
3A55 3900 3852  | door state: 52 1010010
3A55 3900 3852  | door state: 52 1010010
3A55 3900 3852  | door state: 52 1010010
3A55 3900 3852  | door state: 52 1010010
3A55 3900 3852  | door state: 52 1010010
3A55 3900 3852  | door state: 52 1010010
3A55 3900 3852  | door state: 52 1010010
3A55 3900 3852  | door state: 52 1010010

PaulWieland, 2023-11-12:
It seems okay with the wall panel unhooked. I need to find time to do testing with that specific model wall panel and see what it's complaining about.

ylwmamba, 2023-11-12:
Yeah all the functions were working properly without the wall panel. Thanks for taking the time.

rykm48, 2023-11-14:
No I didn’t. I ended up changing it back to 1.0 and the problems went away. The only issue I’m having now is phantom reports of the garage open and then closed. It shows in home assistant log that the garage was open and then followed by closed but never opening or closing. Checking my camera feed logs it never opened so for some reason the ratgdo is reporting it opening and closing on occasion.

PTracks, 2023-11-14:
I'm having the same issue.  Security + 1.0 opener (purple button) with ratgdo v 2.5.  Status constantly switching between "Opening" and "Opened" every few seconds.  Does not seem to be able to control the door although wall button and remotes still function.

I've attached pictures of my opener and wall button.  I can try to get logs tomorrow if that will help.

[image]
[image]
[image]
[image]