PaulWieland/ratgdo #223: Installation Failed - RATGDO 2.5 Board

deep-barot, 2023-11-11:
**IMPORTANT**
Please do not open issues for questions about your order.
If you have a question about your order, reply to the receipt email you received from square.

**Describe the bug**
RATGDO board was detected by computer and then I tried to flash it using the Chrome utility, it fails with "Failed to initialize. Try resetting your device or holding the BOOT button while clicking INSTALL."

**Version information**
* which ratgdo board version? 
   2.5 
* which ratgdo firmware variant and version (MQTT or ESPHome?) 
   ESPHome ratgdo
* which garage door control protocol?
  I have MyQ smart Garage Controller.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
I would expect the board to be flashed with the utility provided on the page.

**Additional context**
Add any other context about the problem here.

cheatoman, 2023-11-11:
This was happening to me. I restarted my computer and then I was able to flash it just fine.

deep-barot, 2023-11-12:
Installed both Chipset drivers from the page before restarting. Restarted the computer and now I have the error that most people are facing. 

[image]

I thought it might be a bad cable as mentioned in the notes, I tried with four different cables. Out of 4, using 3 cables I can see the device in the device manager with the port. So it is not the cable I assume at this point.
[image]

cheatoman, 2023-11-12:
The drivers that worked for me on the end (Windows 10) were the ones I got directly from the FTDI website (the VCP ones). On Windows 10 the initial install would connect to the ratgdo, but flash would fail. It worked after a restart.

Interestingly, on Windows 11 I was getting the "failed to open serial port" error. I did not restart that one and try again though.

deep-barot, 2023-11-12:
Found the solution. Here it is, for anyone that is running around to find the good cable. But it is not the cable, it is the driver.

Here is how I fixed it in Windows 11 with the D1 Mini plugged in:

Pre-requisite:
Download the driver listed above and unzip: https://k2me.tistory.com/145 (Unzip, Zip has another zip in it, unzip that as well)

1. Open "Device Manager"
2. Open "Ports (COM & LPT)"
3. Right-click the troubled device (ex. "USB-SERIAL CH340 (COM10)") and click "Update Driver"
4. Click "Browse my computer for drivers"
5. Click "Let me pick from a list of available drivers on my computer"
6. Click "Have Disk..."
7. Select the unzip folder from Step 1 into the "Copy manufacturer's files from:" and click "OK"
8. Select the model (ex. "USB-SERIAL CH340") and click "Next"
9. Click "Close"

Then go to the the link Paul has given to flash the board, it will work like a charm. Flashed board in a min.

closed 2023-11-12 by deep-barot