PaulWieland/ratgdo #224: Update firmware after flash?

LastToGithub, 2023-11-11:
**IMPORTANT**
Please do not open issues for questions about your order.
If you have a question about your order, reply to the receipt email you received from square.

**Describe the Issue**
Just wondering if we need to update the firmware after we flash the device?

**Version information**
* which ratgdo board version? 2.5
* which ratgdo firmware variant and version (MQTT or ESPHome?) MQTT
* which garage door control protocol? Still working on that one...

PaulWieland, 2023-11-11:
Flashing the device installs whatever firmware you selected which is also the latest available version.

LastToGithub, 2023-11-11:
Thanks!
Is there anyway to check which version of firmware is installed? I just want to check because I'm running into this issue that when I make changes to the ratgo settings and  hit save I get the following message:
Success: rebooting the microcontroller using your credentials.

But then I can't get back to the web GUI. I've waited a few minutes, try power cycling the device. Nothing seems to help.

I can still ping the IP address of the device, however my router's DHCP logs show that the hostname hasn't changed.

When I go back to the flashing page, it seems to indicate the device needs to be flashed again by just giving me the option to install ratgdo or logs & console.

Any suggestions?

PaulWieland, 2023-11-11:
The firmware version is displayed in the header of the web UI and is also printed to the serial console.

Make sure you are leaving this box unchecked:
[image]

It would also help to get a serial log after it reboots. Choose `logs & console`:
[image]
then click `reset device` to reboot it.

LastToGithub, 2023-11-11:
I'm leaving the box unchecked.

Just tried again. After making some very basic changes and updating, I get this page:
[image]

However, I can't get back into the web gui.

When I go back to the flash page and connect, these are the only options I see:
[image]

I'm also not seeing any communication from the device to my MQTT broker.

LastToGithub, 2023-11-11:
Sorry, forgot to mention. The logs are working. Shows "some" changes, for example I renamed the device, however it is still showing the default name of ratgdo-#####.

Still can't browse to the webgui though.
[image]