PaulWieland/ratgdo #229: Ratgdo v2.5 - using “dry” contacts to control door

Juggler00, 2023-11-12:
ratgdo v2.5, standard (MQTT firmware)

I have wired my ratgdo and am able to control my garage door via MQTT. I would like to control my door via the “dry” contacts, optional momentary buttons as per wiring diagram. I am unable to get the door to either open or close via these dry contacts.

The board is powered via the micro-USB port, no connection to +3.3V pin. I have tried with a short wire, “shorting” ground and open, then ground and close. Neither works.

Is there something I need to enable in the firmware?

CrazyCoder, 2023-11-13:
The open/close terminals are for the limit switches so that ratgdo knows the state of the door (open/opening/closing/closed).

ratgdo controls the door using the same Red Ctrl terminal (top left):

[image]

When you set Dry contact mode in the settings, instead of sending the serial commands, it will short Red Ctrl to Ground for 500ms that should trigger the door command like if you pressed a wall button.

Does it make sense?

PaulWieland, 2023-11-13:
The dry contact trigger interface usage depends on the control protocol. If you are using the Dry Contact control for your door, then these are for limit switches to sense the door status.

If you are using Security + 1 or 2, then these are dry contact inputs for triggering the door to open or close (requires a 500ms contact closure).

Juggler00, 2023-11-14:
Thank you for your response and clarification. I am using the Security +2.0 protocol, but was hoping to not have to connect the device to WiFi.

I have done further testing and I seem to only be able to activate the dry contacts sporadically. I have verified the connections and tested continuity. I'm wondering if there is a bug with this? I have attached my log in the hopes it might shed some light. I found that I sometimes had to trigger the "close" to open the door and vice versa.

I appreciate my use-case is somewhat unique, but if you have a moment would appreciate your thoughts. Happy to do any testing.
```
detected
rolling code for 34624 391|reboot2 : 5501009229A4D269A4D26828379C5921528B24
5501009229A4D269A4D26828379C5921528B24
550100293539A6A04982218A349A5F0F724F71 | STATUS: door:2 light:1 lock:0 obs:1
550100293539A6A04982218A349A5F0F724F71 | STATUS: door:2 light:1 lock:0 obs:1
Dry Contact: close the door
The door is already closed
Dry Contact: open the door
rolling code for 34624 392|door1 : 5501001904120400004807960D6737DA8DF7F3
rolling code for 34624 392|door2 : 5501001904120400004806960D6737DA8DE7F3
rolling write successful
Dry Contact: close the door
The door is already closed
Dry Contact: close the door
The door is already closed
5501001904120400004807960D6737DA8DF7F3
5501001904120400004806960D6737DA8DE7F3
5501009422F29A5736BEF72924B2031040C303 | STATUS: door:4 light:1 lock:0 obs:1
Door state opening
QUEUE MSG SENT [ratgdo-inner_door/status/door] 
opening
550100981D904EA7120124A63CFF8AFA9F7611 | MOTION:1
Motion state detected
QUEUE MSG SENT [ratgdo-inner_door/status/motion] 
detected
rolling code for 34624 393|reboot2 : 550100182490492492012688028BCD27725D24
550100182490492492012688028BCD27725D24
5501009422F29A5736BEF72924B2031040C303 | STATUS: door:4 light:1 lock:0 obs:1
5501009230A9F7E1EDC2611A3BFFD3E19EB34E | STATUS: door:1 light:1 lock:0 obs:1
Door state open
QUEUE MSG SENT [ratgdo-inner_door/status/door] 
open
550100981D904EA7120124A63CFF8AFA9F7611 | MOTION:1
Motion state detected
QUEUE MSG SENT [ratgdo-inner_door/status/motion] 
detected
rolling code for 34624 394|reboot2 : 55010082192C964B649258260147369FAD7693
55010082192C964B649258260147369FAD7693
550100610BC4081E12249E09269A0508300407 | STATUS: door:1 light:1 lock:0 obs:1
550100981D904EA7120124A63CFF8AFA9F7611 | MOTION:1
Motion state detected
QUEUE MSG SENT [ratgdo-inner_door/status/motion] 
detected
rolling code for 34624 395|reboot2 : 5501008110492082092491182D954122F22124
5501008110492082092491182D954122F22124
550100610BC4081E12249E09269A0508300407 | STATUS: door:1 light:1 lock:0 obs:1
550100981D904EA7120124A63CFF8AFA9F7611 | MOTION:1
Motion state detected
QUEUE MSG SENT [ratgdo-inner_door/status/motion] 
detected
rolling code for 34624 396|reboot2 : 550100503EDB6DB7DF7DB4061262B65FCDF64B
550100503EDB6DB7DF7DB4061262B65FCDF64B
550100610BC4081E12249E09269A0508300407 | STATUS: door:1 light:1 lock:0 obs:1
550100981D904EA7120124A63CFF8AFA9F7611 | MOTION:1
Motion state detected
QUEUE MSG SENT [ratgdo-inner_door/status/motion] 
detected
rolling code for 34624 397|reboot2 : 5501004A2DBFDF6FBEDF7CA836B8CB21728B24
5501004A2DBFDF6FBEDF7CA836B8CB21728B24
550100610BC4081E12249E09269A0508300407 | STATUS: door:1 light:1 lock:0 obs:1
550100981D904EA7120124A63CFF8AFA9F7611 | MOTION:1
Motion state detected
QUEUE MSG SENT [ratgdo-inner_door/status/motion] 
detected
rolling code for 34624 398|reboot2 : 550100A52596CB25965B2856377C3699A98693
550100A52596CB25965B2856377C3699A98693
550100610BC4081E12249E09269A0508300407 | STATUS: door:1 light:1 lock:0 obs:1
550100610BC4081E12249E09269A0508300407 | STATUS: door:1 light:1 lock:0 obs:1
Dry Contact: open the door
The door is already open
550100981D904EA7120124A63CFF8AFA9F7611 | MOTION:1
Motion state detected
QUEUE MSG SENT [ratgdo-inner_door/status/motion] 
detected
rolling code for 34624 399|reboot2 : 550100A43EFBEFB7DF7FBB48108BCD273A5D24
550100A43EFBEFB7DF7FBB48108BCD273A5D24
550100610BC4081E12249E09269A0508300407 | STATUS: door:1 light:1 lock:0 obs:1
550100981D904EA7120124A63CFF8AFA9F7611 | MOTION:1
Motion state detected
QUEUE MSG SENT [ratgdo-inner_door/status/motion] 
detected
rolling code for 34624 400|reboot2 : 550100122926924924D26B260947B6DFAD7693
550100122926924924D26B260947B6DFAD7693
Dry Contact: close the door
rolling code for 34624 401|door1 : 5501001112CB20924964951829974822F20134
rolling code for 34624 401|door2 : 5501001112CB20924964941829974822F22134
```