PaulWieland/ratgdo #233: Not booting

ibtvt, 2023-11-13:
I recently received my order, but when I plug the device in the LED turns on for a split second then turns back off.  I've tried multiple known good micro USB cables, USB power bricks and computers.  None of the computers see the device as connected, and I confirmed the drivers are installed.  I removed the D1 Mini from the ratdgo board and get the same result.

AdamSutch, 2023-11-15:
I'm getting the same thing on 1 of the 2 I just got today.
The first one, I plugged in, the blue LED came on and stayed on, I was able to flash the ESP version, installed it and everything is working fine.
The second one, the blue LED blinks twice when I plug it in but I'm not able to install anything on it. I also noticed it's broadcasting an open wifi called "AI-THINKER_xxxxxx" (assuming the x are the mac address, so omitted).
Tried multiple computers including Mac and Windows, Chrome and Edge and 4 different known to be working USB cables (included the one that worked with the other board).

I don't think it's dead but it's probably pre flashed with something that's not allowing it to connect over serial.

AdamSutch, 2023-11-15:
Not sure if it'll help you @ibtvt or not, but I managed to get mine flashed.
Had to ground D3 (4th pin on the "right") to the metal case while plugging it in (apparently this enables flash mode), then I was able to take a .bin file I generated with ESP home and install it with https://github.com/esphome/esphome-flasher/releases/tag/1.4.0 (the desktop app, none of the web installers worked).

This diagram might help https://preview.redd.it/33u1orct8zva1.png?width=3294&format=png&auto=webp&s=d19d3cdb171b3b5d6b156aa4dc98015baa0093fd

And I only needed D3 to be touching the metal case (I guess it's the same as ground) for a second while I plugged it in, I didn't need to hold it for any time.

From what I gather, the chip I got (and probably you too) come pre-installed with some software that prevents the USB from being able to start flash mode like the normal ESP8266's.

Hope it helps. GL;HF

PaulWieland, 2023-11-15:
@ibtvt please reply to your square receipt with your issue

closed 2023-11-15 by PaulWieland