PaulWieland/ratgdo #234: General pre-sales questions

NukeThemTillTheyGlow, 2023-11-13:
**IMPORTANT**
Please do not open issues for questions about your order.
If you have a question about your order, reply to the receipt email you received from square.

**Describe the Issue**
I have a few basic questions before ordering

1. Any place to place a power tap on the opener to avoid a separate power adapter?	 
2. If not, I am looking to go with some basic 5v Amazon Basics power brick and cables, anyone have any problems with that?
3. Looking to use Cat5e wiring from Ratgdo to/from opener, any issues with that?
4. I assume update the firmware/software first, then test wifi connection on the ground before wiring for each one of them?				

Include a photo of your wiring!

**Version information**
* which ratgdo board version?   From the mix of Red and Yellow buttons, v2.5
* which ratgdo firmware variant and version (MQTT or ESPHome?)  Home Assistant
* which garage door control protocol?  Two are 2.0, one must be 1.0 (just says Security +, its the Red button unit)

**Additional context**
Add any other context about the problem here.

kramttocs, 2023-11-14:
1. Depends upon the opener. Paul has some photos for the v1 board you _might_ be able to use. Personally, I would only power from the opener itself by hooking it to the battery circuit (if present and voltage will need adjusted). Otherwise, as Paul mentions, hard to say if the board can handle the added load.
2. That's fine
3. That's fine and what I did as well. Actually, in my case, all of my obstruction sensor wiring is in Cat6 and the sensors themselves plug into an ethernet jack in the wall. Makes them easier to replace when a raccoon gets trapped in there and yanks on everything they can get their grubby mitts on.
4. Yes as the update will actually prompt you to setup wifi when finished so might as well do it all at the same time. Then move it to the opener.

NukeThemTillTheyGlow, 2023-11-15:
Ordered 3 2.5 boards last night.   Picking up power bricks and usb cables soon.   will wire to operers using cat5e wire.
For install with Home Assistant is there a preferred option...MQTT or ESPHOME?  Any potential conflicts with other integrations I should be aware of?

closed 2023-11-16 by NukeThemTillTheyGlow