PaulWieland/ratgdo #236: Rolling code not synced?

webmogul1, 2023-11-14:
Everything seems to be installed correctly.  
Entities are being reported correctly in Homeassistant but I can't control anything.

ESPHOME version.
Board v2.5

`[23:07:43][D][light:036]: 'Light' Setting:
[23:07:43][D][light:047]:   State: ON
[23:07:43][D][number:012]: 'Rolling code counter': Sending state 28.000000
[23:07:45][D][esp8266.preferences:238]: Saving preferences to flash...
[23:07:45][W][component:214]: Component preferences took a long time for an operation (0.06 s).
[23:07:45][W][component:215]: Components should block for at most 20-30ms.
[23:07:47][D][light:036]: 'Light' Setting:
[23:07:47][D][light:047]:   State: OFF
[23:07:48][D][number:012]: 'Rolling code counter': Sending state 29.000000
[23:07:50][D][esp8266.preferences:238]: Saving preferences to flash...
[23:10:06][D][ratgdo:210]: Status: door=CLOSED light=OFF lock=UNLOCKED`

mattvirlee, 2023-11-14:
Low hanging fruit - did you unplug the garage door opener first before plugging in the ratgdo board? I think I had the same issue as you when I didn’t install in that order.

webmogul1, 2023-11-14:
Well, sort of. 
I installed the firmware and then played with the controls a little before connecting to the garage door opener.

If that was the problem, what do I have to do now?

PaulWieland, 2023-11-14:
Please fill out the issue template when creating a new issue. You have to include a photo of the board and wiring and an explanation of what versions and control protocols you are using..

mattvirlee, 2023-11-14:
Disclosure: not sure if this is how Paul would solve it, but here’s how I did:
- Unplug the Opener
- Reflash the Board
- Reconfigure the board
- Connect to opener wiring if not already
- Power the Ratgdo
- Power the Garage door
- Cycle the door using the physical wall button
- Wait a moment or seven while the ratgdo syncs

That got me sorted.

PaulWieland, 2023-11-14:
Yeah, none of that should be needed :)

@webmogul1  please post the requested info...

webmogul1, 2023-11-14:
> Disclosure: not sure if this is how Paul would solve it, but here’s how I did. Might be a “hammer” approach when there is a more surgical one, but…..
> 
> * Unplug the Opener
> * Reflash the Board
> * Reconfigure the board
> * Connect to opener wiring if not already
> * Power the Ratgdo
> * Power the Garage door
> * Cycle the door using the physical wall button
> * Wait a moment or seven while the ratgdo syncs
> 
> That got me sorted.

That did the trick! Thanks!

webmogul1, 2023-11-14:
> Yeah, none of that should be needed :)
> 
> @webmogul1 please post the requested info...

Maybe the issue was because I played with the commands in Homeassistant before connecting to the GDO?

webmogul1, 2023-11-14:
MyQ App just got deleted! 
Thanks @PaulWieland !

closed 2023-11-14 by PaulWieland