PaulWieland/ratgdo #239: Received only payment receipt - no order confirmation receipt

scooterusa, 2023-11-15:
I placed an order on October 31st but never received a confirmation email with an order number.

I also checked my spam folders and all other folders. So I didn't miss it if it got sent. I waited a little bit before reaching out here in case the order confirmation email got delayed a little 

Who do I contact to have the order confirmation sent, or how can I have it sent?

Replying to the email in the payment receipt does not work, and I see no contact information I can use for this purpose.

PaulWieland, 2023-11-15:
Please reply to your square receipt

closed 2023-11-15 by PaulWieland

scooterusa, 2023-11-15:
Just did... Hopefully this time it works.

PaulWieland, 2023-11-15:
I didn’t receive anything. Send an email to ***@***.***

scooterusa, 2023-11-16:
I just sent it again and made absolutely sure it was sent to an email address. I used the one that shows up when I go to reply to the payment receipt email. For good measures I sent it twice... One from the email address that I used when placing the order, and again from my personal domain name.

Email address after the @ is "reply2.squareup.com"

donavanbecker, 2023-11-16:
I have tried to email the receipt directly also and no response.

wastingaeons, 2023-11-16:
experiencing this as well

scooterusa, 2023-11-16:
Just wanted to see if you got my yet PaulWieland. 

Thanks for helping me with getting the order confirmation receipt sent again.

donavanbecker, 2023-11-16:
> I have tried to email the receipt directly also and no response.

got my update email today. but no response.