PaulWieland/ratgdo #240: Security+ 1.0 wall button LEDs flicker after ratgdo install

dlgoodr, 2023-11-15:
**IMPORTANT**

**A photo of your ratgdo board and wiring is required.**

Please do not open issues for questions about your order.
If you have a question about your order, reply to the receipt email you received from square.

### Describe the Bug
A clear and concise description of what the bug is. 

### Version information
* ratgdo board version: [1.0, 2.0, 2.5]
* ratgdo firmware variant: [MQTT, ESPHome]
* control protocol: [Sec+2, Sec+1, Dry Contact]

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

### Additional context
Add any other context about the problem here.

```
paste ratgdo serial console log here
```

blulite, 2023-11-15:
Me too.   From logs:

 ```
Looking for security+ 1.0 wall panel…
Looking for security+ 1.0 wall panel…
Looking for security+ 1.0 wall panel…
Looking for security+ 1.0 wall panel…
Looking for security+ 1.0 wall panel…
Looking for security+ 1.0 wall panel…
Looking for security+ 1.0 wall panel…
No wall panel detected. Switching to emulation mode.
35FF 35FF 35FF 3533 3353 38FF 3A3A Light state off
```

It seems to work (both MQTT commands and the wall panel Open/Close).  But the wall panel light control does not work.

PaulWieland, 2023-11-15:
Please post a photo of your board and wiring

dlgoodr, 2023-11-15:
Here you go. Brief video of the flickering at the wall button, too: https://imgur.com/a/Q0Ulw5u

I don't think I have the same problem as @blulite, I'm talking about the LED behind the button on the wall opener, not the bulb inside the motor unit. (Though mine does report spurious lights-on and lights-off events)

dlgoodr, 2023-11-15:
Hm, never mind, I might have the same problem as @blulite now that I've connected to the device to pull logs.

```
Looking for security+ 1.0 wall panel...
Looking for security+ 1.0 wall panel...
Looking for security+ 1.0 wall panel...
Looking for security+ 1.0 wall panel...
Looking for security+ 1.0 wall panel...
No wall panel detected. Switching to emulation mode.
3535 3533 3353 38FF 3A3A Light state off
QUEUE MSG SENT [ratgdo/little_garage/status/light] 
off
3A39 383A  | door state: 3A 111010
Door state open
QUEUE MSG SENT [ratgdo/little_garage/status/door] 
open
383A  | door state: 3A 111010
3938 3A39 383A  | door state: 3A 111010
39FF 383A  | door state: 3A 111010
3938 3A39 3855  | door state: 55 1010101
Door state closed
QUEUE MSG SENT [ratgdo/little_garage/status/door] 
closed
3A59 3900 3855  | door state: 55 1010101
3A59 3900 3855  | door state: 55 1010101
3A59 3900 3855  | door state: 55 1010101
3A59 3900 3855  | door state: 55 1010101
3A59 3900 3855  | door state: 55 1010101
```
the door was not opened or closed in reality during this time.

blulite, 2023-11-15:
I don't own an official unit yet, so I am not sure that photos of the wiring will help.  I built one as I await shipment of the genuine unit from your backlog.   If you prefer to work with me once I install in the real one, that's ok.  But I do have the same issues as above, flicker at the wall unit LED.    

Everything is working fine, with the exception of the flicker and the wall unit "light" button does not work.  And the light does come on by itself randomly (it's sensed by RATDGO and sent as "ON" via MQTT).  I can turn it off via MQTT, but it turns itself back on in a bit.

My GDO model is: Chamberlain 2245
Wall Panel is:  [041A5273-1](https://www.liftmaster.com/wall-control-panel/p/041A5273-1)

kkohls, 2023-11-16:
I recently installed two version 2.5 ratgdo boards on my two Chamberlin Security + 1.0 openers (purple button) and am experiencing the same thing. I have the same push button panels in dlgoodr’s post. 

-The led in the push button rapidly flickers
-The light button doesn’t work
-I get occasional random open/close mqqt messages
-I will also sometimes get double mqqt messages when actually opening/closing the door.

vtylerbaker, 2023-11-16:
Checking in here to report the same behavior plus another oddity.  Using 2.5 boards and Security+ 1.0 with Red Button opener.

I recently installed two 2.5 boards.  Both installs were flawless and detected by mqtt/HA.  I can open/close both doors via HA and the wall panels.  I am wiring directly to the opener vs. using the passthrough ports.  I am also experiencing the wall panel flicker and unable to control the light from the wall panel.  The light for one of my boards will turn on randomly.  I can shut the light off via HA (but not the wall panel) and it will turn back on consistently after an exact amount of time, but that amount of time isn't consistent after reboots of the board.  For instance, I noticed it was exactly 66 seconds that the light would turn back on after turning it off in HA.  It would repeat indefinitely.  After a reboot, that internal changed to 10 seconds.  

  In addition, when the boards are plugged in and operational I randomly lose the ability to use the openers in my vehicles.  It's not consistent, but happens very often.  Behavior all returns to normal without power to the boards.

joltman80, 2023-11-16:
> In addition, when the boards are plugged in and operational I randomly lose the ability to use the openers in my vehicles. It's not consistent, but happens very often. Behavior all returns to normal without power to the boards.

As to the HomeLink not working in your vehicles, I had this issue when I used a Shelly device and a spare 12V adapter.  Sometimes, I'd have to press the HL button in my vehicle 10+ times to get the GD to open.  Once I unplugged that 12V adapter, the issue immediately went away and the GD has been opening successfully on the first button push.  What PSU do you have supplying power to the ratgdo?

vtylerbaker, 2023-11-16:
> What PSU do you have supplying power to the ratgdo?

Great question - and something I forgot to mention/ask.  I installed a [multi-outlet extender](https://www.amazon.com/dp/B08R6S1M1K?psc=1&ref=ppx_yo2ov_dt_b_product_details) on my duplex outlet in the garage.  Basically it just provides more plugs and adds USB ports.  Those ports are 5V/4.8A and there are three of them.  I think it's 5V total and I do have a Wyze camera plugged into one of the other ports.  Was wondering if power consumption may be playing a part here as well.

Doughboy68, 2023-11-20:
I have a 1.0 opener with purple button. Was considering ordering one if these but think I’ll hold off now based on this issue until resolved. I currently have a meros connected and works well - the security+ 1.0 you can just short the wires and opens closes door. The reason I was looking at a ratgdo is because it can detect obstruction, control light, and can set open or close to a percentage using esphome but is this supported in 1.0 and mqtt?  ESPHome  version is not working yet for 1.0 as I understand ?

PaulWieland, 2023-11-20:
These wall panels are "dumb" in that they don't query or receive information. They only send commands. the light flickering is normal because ratgdo is polling the GDO on the serial line and that traffic causes the light to flicker.

This is all a consequence of the emulation mode ratgdo uses to get full bi directional communication with sec+ 1.0 openers. The same thing would happen if you hooked up an 889LM in parallel with this older style panel.

closed 2023-11-20 by PaulWieland