PaulWieland/ratgdo #242: Question related to a case i purchased on ETSY

DavidRSullivan, 2023-11-15:
Hi, 

I purchased a Ratgdo a week ago for my RJ070, it's on backorder.  I found a case on ETSY:  https://www.etsy.com/listing/1590980806/ratgdo-v25-pcb-case?ref=hp_active-mission-recs-rv-1-1

I just received the following message from them: "It has come to our attention that some people who have ordered a v2.5 version of the ratgdo board have started to receive v2.5i instead. To prevent you from incurring additional costs if you receive a different board than what you purchased, we are placing all open v2.5 orders on hold until we can confirm which version of the board you have received."

Is there a version 2.5i board and is it a different footprint?

Thank you for all your work on this product, I hope to hear back soon.

foreverimagining, 2023-11-15:
Yes, there was another sudden hardware change in the last few days. The v2.5i is supposed to do the same thing as the v2.5, but it has the ESP integrated into the board and a very different footprint.
[image]
[image]

Unless @PaulWieland is able to provide the necessary measurements, @samwiseg0 will have to wait for someone from the community to happen to receive the new board and have an interest in a 3D printed case. Personally, I sincerely hope that my board will still be original v2.5 because I printed the case soon after receiving my tracking number a week and a half ago, a couple days before I learned of the hardware change. Hopefully, I'll be finding out today.

PaulWieland, 2023-11-15:
As stated in the new issue guideline: If you have a question about your order, reply to your square receipt.

foreverimagining, 2023-11-15:
@samwiseg0 Oh, good! It's supposed to be out for delivery, so if you don't hear from me later today, I did get original v2.5. I'm planning to post my make on Printables in a few days after I get it installed.