PaulWieland/ratgdo #244: openhab configuration example?

xanderphillips, 2023-11-15:
Will be connecting to openHAB.
I have not seen any examples for configuring it in that system and am hoping someone has set it up before and can share their config for me to base mine off of.

### Version information
ratgdo board version:  2.5
ratgdo firmware variant: [MQTT]
control protocol: [Sec+2]

### Additional context

jshatch, 2023-11-16:
You will need to use the MQTT binding and Homeassistant discovery.  It should pick it up automatically if your broker is set up and both openHAB and ratgdo can talk to it (ie your aclfile is configured properly if you are using it).  You'll probably get better discovery results if you use the ESPHome firmware and configure it to use MQTT.

xanderphillips, 2023-11-16:
I figured I'd need to wire it through MQTT, just was hoping someone that had already done it had configs I could reference.
I've wired up other things blindly through MQTT before so won't be a complete blocker for me, was just hoping to leverage other's learnings.

closed 2023-11-16 by xanderphillips

computergeek1507, 2023-11-20:
Thing topic RatgdoGarageDoor "Ratgdo Garage Door" {
      Channels:
          Type switch : switch "Door Switch" [
            commandTopic="ratgdo/ratgdo-55451/command/door",
            on="open",
            off="close"
          ]
          Type contact : door "Door Status" [
            stateTopic="ratgdo/ratgdo-55451/status",
            on="open",
            off="close"
          ]
          Type switch : light "Light Status" [
            stateTopic="ratgdo/ratgdo-55451/status/light",
            commandTopic="ratgdo/ratgdo-55451/command/light",
            on="on",
            off="off"
          ]
          Type string : obstruction "Obstruction" [
            stateTopic="ratgdo/ratgdo-55451/status/obstruction"
          ]
          Type string : status "status" [
            stateTopic="ratgdo/ratgdo-55451/status/door"
          ]
          Type string : availability "Availability" [
            stateTopic="ratgdo/ratgdo-55451/status/availability"
          ]
          Type string : motion "Motion" [
            stateTopic="ratgdo/ratgdo-55451/status/motion"
          ]
    }