PaulWieland/ratgdo #245: Multiple Messages [ratdgo:467] Collision detected, waiting to send packet

wesf99, 2023-11-15:
I am using Homeasssistant ESPHome to connect and configure my Ratdgo 2.5 device.

It seems to have setup with the decvice and entities appearing in Home Assistant.

However the log looks like the device is having transmission errors and repeatedly displaying multiple Messages:
                       [ratdgo:467] Collision detected, waiting to send packet.

I have not connected the wiring yet as I am concerned about this error message.

### Version information
* ratgdo board version: [2.5]
* ratgdo firmware variant: [ESPHome]
* control protocol: [noy configured yet]

**To Reproduce**
Power on Ratgdo and look at logs.

**Expected behavior**
I was not excpecting error messages.

### Additional context
I am new to these types of control boards, ESPHome and MQTT. I have probably made some mistake during the setup/configuartion

```
is this the serial console:
![image](https://github.com/PaulWieland/ratgdo/assets/55278725/cd58a882-8055-457e-a01b-5038645f866b)

please find attached log:
[logs_ratgdov25-8e63d1_run (1).txt](https://github.com/PaulWieland/ratgdo/files/13370733/logs_ratgdov25-8e63d1_run.1.txt)

```

PaulWieland, 2023-11-15:
Please post a photo of your ratgdo board and wiring.

wesf99, 2023-11-16:
Please see attached photos
[image]
[image]

wesf99, 2023-11-16:
I have a MYQ MS65MYQ door opener. My Wall Mount Controll is wireless and hence not connected to the opener by wires. I cabled the ratgdo to the opener by running a wirefrom Red Ctrl to Teminal 1 Push Button and White Gnd to Terminal 2 Ground. these two terminals will act as Dry Contcat when connected and open the door. While the Collision Detection messages have stopped, the Garge door does not open or close. ANy ideas what I can try next. Please see attached files,
MyQ MS65MYQ Control Panel
[image]
Ratdgo Connected
[image]
Home Assistant Log
[logs_ratgdov25-8e63d1_logs.txt](https://github.com/PaulWieland/ratgdo/files/13375998/logs_ratgdov25-8e63d1_logs.txt)

limwakra, 2023-11-27:
Solution is to connect to 0 (e-serial port) instead of 1.