PaulWieland/ratgdo #246: Door status unknown (ratgdo 2.51 sec +1)

BeebleZap, 2023-11-15:
Using MQTT version 2.51 sec+1

I can open/close the door, and turn on/off the light using MQTT controls in Home Assistant.
The light status is reported correctly, but not the door status.

So if the door is open, clicking any button (open or close) will close the door, and vice-versa.

[image]
[esp-web-tools-logs.txt](https://github.com/PaulWieland/ratgdo/files/13370772/esp-web-tools-logs.txt)

BeebleZap, 2023-11-16:
Liftmaster model 3585CS

dbrewer-wdc, 2023-11-20:
@BeebleZap, this is a problem that I have too. I've posted over on ratgdo-mqtt because that's the mode I'm running in. My understanding is that the wall switch requests status from the garage door opener and the ratgdo captures it and then reports status to the use through ESPHome or MQTT. However, in my case, and maybe your case, the wall switch does not ask for status, so you will never get it. There is a way to put the ratgdo in emulation mode and that works, but I'm not sure it is the best thing to do. Basically the ratgdo asks for status and that seems to work. 

My solution right now is that I have another device, an ESP32 in the garage being used for something else but had a free IO on it. I wired it to a magnetic reed switch on the garage door. I'm using MQTT on both devices so I have the  ESP32 over-write the status space the ratgdo uses and Homeassistant just picks it up and presents the correct status. Works perfectly.

Probably the best solution is to have an option with the ratgdo code to allow for both sec+1 AND an input for a reed switch. Another option would be to use the emulation mode and get rid of the wall switch. Wire up your up down buttons to be used for local control instead.